ANYbotics/elevation_mapping #153: Is that any configuration for realsense L515?

altfool, 2021-01-07:
Hi, 
I'm rencently working on combing the elevation mapping with intel realsense L515. I checked the configuration in elevation_mapping directory and found the configuration for ZR300 and D435. I planned to copy D435's configuration to create a configuration for L515, but after I checked [L515 configuration online](https://www.intelrealsense.com/lidar-camera-l515/), I can't find something similar to the normal_factor_*, so would you mind telling how to get those parameters or do you have a configuration for the realsense L515?

maximilianwulf, 2021-01-07:
Hey @altfool, there is currently no configuration for the L515 available in this repository. You can derive it yourself with the help of the available papers. In case you manage to figure out a configuration, it would be great if you could make a PR.

Also, since the L515 is actually a Lidar and not a stereo camera, you should look into the other Lidar configurations, such as this one: https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/config/sensor_processors/velodyne_HDL-32E.yaml. This is also the reason why there are no `normal_factor_*` parameters given.

altfool, 2021-01-07:
> Hey @altfool, there is currently no configuration for the L515 available in this repository. You can derive it yourself with the help of the available papers. In case you manage to figure out a [...]

Thank you very much.

usamex, 2021-01-13:
Hey, sorry to invoke this issue but how can I configure a structured light sensor? I have D455 with me but I don't know how can I adjust normal_factor_* parameters. Is there a paper for it?

closed 2021-01-13 by maximilianwulf