ANYbotics/elevation_mapping #155: D435i issue

kiwi-sherbet, 2021-02-05:
Hi, I would like to use `elevation _mapping` package with Realsense d435i. When I ran it, I got the following error messages.

`Parameter 'point_cloud_topic' is deprecated, please use 'input_sources' instead`

It seems that the configuration for the parameter `input_sources` is not set properly, and I also found the related terms in the readme file. But, I do not know how I can change `input_sources` parameter. Where can I set up the configuration for it?

Thanks!

maximilianwulf, 2021-02-08:
Hey @kiwi-sherbet, 

you can configure `input_source` with : 

```
input_sources:
  front: # A name to identify the input source
    type: pointcloud # Supported types: pointcloud
```

kiwi-sherbet, 2021-02-08:
Thank you for answer!

Where (in what file) can I change the lines for configuration?

maximilianwulf, 2021-02-08:
Hi @kiwi-sherbet, you can add these changes to your configuration yaml file.

kiwi-sherbet, 2021-02-09:
Thank you for reply!
Still, I get an error that "Could not configure input source front because no sensor_processor was given." It seems that I am missing some configuration setups.
If there is any demo or example for D435i, can I ask for it? It would be great help!

maximilianwulf, 2021-02-09:
Sorry, forgot that. You also have to add the sensor processor, like: 

```
input_sources:
 camera: # A name to identify the input source
    type: pointcloud # Supported types: pointcloud
    topic: /your_topic # your topic
    queue_size: 1
    publish_on_update: true
    sensor_processor: 
      ignore_points_above: .inf
      ignore_points_below: -.inf
      cutoff_min_depth: 0.1
      cutoff_max_depth: 1.5
      type: structured_light # https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8768489
      normal_factor_a: 0.001063
      normal_factor_b:  0.003949
      normal_factor_c: 0.0
      normal_factor_d: 0.0008278
      normal_factor_e: 1
      lateral_factor: 0.01576 # approx 1 deg
```

kiwi-sherbet, 2021-02-09:
Thank you for information! Can I ask one more question about how to configure IMU data? 
I still have the following error.
[ERROR] [-]: Could not get pose information from robot for time 1612910006.554518. Buffer empty?

Thank you!

maximilianwulf, 2021-02-11:
Hey, you need to have working state estimation that outputs a pose with a covariance. Do you have such a system running? 
Elevation mapping does not accept raw IMU measurements.

usamex, 2021-02-12:
You have to remap your _Odometry_ type data to _PoseWithCovarianceStamped_ type. You can do it with a simple script like this:
```
#!/usr/bin/env python
import rospy
from nav_msgs.msg import Odometry
from geometry_msgs.msg import PoseWithCovarianceStamped

def odom_callback(msg, arg):
    publisher = arg
    new_message = PoseWithCovarianceStamped()
    new_message.header = msg.header
    new_message.pose = msg.pose
    publisher.publish(new_message)
if __name__ == '__main__':
    rospy.init_node('odom_to_pose')

    topic_name = rospy.get_param('~odom_topic', '')
    publisher_name = rospy.get_param('~pub_topic_name', '')
    if topic_name == '' or publisher_name == '':
        rospy.logerr("Couldn't get subscriber's topic name or publisher's topic name . Exiting...")
    else:
        publisher = rospy.Publisher(publisher_name, PoseWithCovarianceStamped, queue_size=10)
        rospy.Subscriber(topic_name, Odometry, odom_callback, publisher)
        while not rospy.is_shutdown():
            pass
```

maximilianwulf, 2021-03-07:
Hey @kiwi-sherbet, it seems elevation mapping does not get the input pose. You would get a message saying `Received first pose`. Can you double-check that you actually get the pose out of your slam module?

kiwi-sherbet, 2021-03-15:
Hi @maximilianwulf , I double-checked the input pose in `PoseWithCovarianceStamped`, as @usamex mentioned. I am currently checking `/tf` topic and configuration at `simple_demo_robot.yaml` but still got the same results.

askokostic, 2021-03-30:
I believe I'm having a similar issue, although my setup is slightly different (T265 for odometry + D455 for depth).

maximilianwulf, 2021-05-31:
Hi @kiwi-sherbet, did you manage to make it work?

kiwi-sherbet, 2021-05-31:
@maximilianwulf Oh, yes. Thank you!

closed 2021-05-31 by kiwi-sherbet

maximilianwulf, 2021-05-31:
Great to hear.

wxk514, 2021-11-29:
> 
Have you solved this problem? I met the same problem and followed the above steps. Could you tell me how to solve this problem.
[ERROR] [-]: Could not get pose information from robot for time 1612910006.554518. Buffer empty?

wxk514, 2021-11-29:
> PoseWithCovarianceStamped

How to set  time stamps?thanks!

manavthakkar, 2024-05-28:
> > PoseWithCovarianceStamped
> 
> How to set time stamps?thanks!

did you figure out how to set time stamps?