ANYbotics/elevation_mapping #156: Handling of second floor holes

blablebli-robots, 2021-02-18:
Hello there!

How do you guys handle holes on a second floor? I am using a LiDAR with elevation mapping, but the laser returns going through the hole on the floor are outside the elevation map, so the map has missing data (filled with nan) in the area corresponding to the hole.
Any suggestions?
I'm trying to navigate unknown terrain, so as the robot moves more nans are added to the map (as the map scrolls) and it becomes hard to distinguish which nan correspond to the actual hole and which to new, yet to fill data.

Thanks!

usamex, 2021-02-25:
LIDARs don't produce dense data and I think because of that you are encountering this problem. It would be better for you if you would go with a different setup (for instance, you can use, Structured Light (Realsense) or TOF sensors) or you can use a motor under the LIDAR to rotate it so it can make your pointcloud data denser.

blablebli-robots, 2021-02-25:
Thank you for you response @usamex, I am using a tilting LiDAR, it produces a dense enough cloud for elevation map to work nicely. However if the robot is on a second floor, and there is a hole on the floor, like on the edge of a mezzanine, the floor before the edge will be registered correctly, but as the the laser sweeps, the returns corresponding to the laser beams going through the hole are outside the local map (because of height they go too far), so there is no data registered on the map (filled with nan) on cells corresponding to those returns. Any ideas?

maximilianwulf, 2021-03-07:
Hi @blablebli-robots, 

> the returns corresponding to the laser beams going through the hole are outside the local map (because of height they go too far), so there is no data registered on the map (filled with nan) on cells corresponding to those returns. Any ideas?

Increasing the map size? But as elevation mapping is a robot centric package increasing the size is only useful to a certain extend. 

Also, it is also intended that the holes are saved as NaN indicating that this area should be explored or prevented.

blablebli-robots, 2021-03-08:
Thank you @maximilianwulf, I'll think of some logic to distinguish between to be explored and already scanned regions.

maximilianwulf, 2021-03-08:
Hi @blablebli-robots, have fun. Please also make sure to check out the traversability estimation package from RSL. There might be useful information for you. https://github.com/leggedrobotics/traversability_estimation