ANYbotics/elevation_mapping #158: problems with using RealSense D435i

mark9714, 2021-04-14:
I got a rosbag recorded from RealSense D435i and slamware radar, "rostopic list" like this.
[image]

And this is the launch file modified by me.
[image]

I use the odom_to_pose.py to change the Odometry to PoseWithCovarianceStamped.
[image]

And here is my configure file "my.yaml".
[image]

However when I use "roslaunch elevation_mapping_demos simple_demo.launch" to start the launch file,I got some information like this.
[image]

And then it turns out to like this.
[image]

And I got nothing in Rviz.
[image]

Does anyone have any idea about it?

maximilianwulf, 2021-05-16:
Hey, your tf data does not seem to be correct. How are you providing the tf information of the robot body? The provided pose seems to be fine.

naoya-kumagai, 2021-09-11:
@mark9714 
Did you figure this out? I'm trying to do something similar, and I'm interested.

mark9714, 2021-09-12:
yes，but I'm Chinese.My English isn't very well,I don't know how to express it.

naoya-kumagai, 2021-09-12:
@mark9714 
If you can post your working code that would be great!