ANYbotics/elevation_mapping #159: Waffle demo not working correctly

fus3, 2021-04-24:
Hello, im currently working on using the elevation_mapping package with T265 and D435 for my quadruped robot on a Jetson Nano running 18.04 and ROS Melodic. Because it is a Jetson nano im forced on using ROS Melodic, I Know that exists a 20.04 version but its not official supported and i prefer use melodic since its more stable anyways because im using melodic and the elevation_mapping uses Newer versions of PCL which 18.04 doesnt work, after a lot of tries of build and error i solved some problems like:
 - https://github.com/ANYbotics/elevation_mapping/issues/151
(because of using version 1.8.1 of PCL i did change PCL_MAKE_ALIGNED_OPERATOR_NEW for EIGEN_MAKE_ALIGNED_OPERATOR_NEW on the file.
 - https://github.com/PointCloudLibrary/pcl/pull/4266
 - Also Jetson did have a error with internal compiler (program cc1plus) which resulted on low memory: using catkin_make -j2 -DCMAKE_BUILD_TYPE=Release did it solve
 - Also i did https://github.com/ANYbotics/grid_map/issues/292
 Ok, so after all this, the build was fine, running the grid map examples work perfectly, when running the  rostest elevation_mapping elevation_mapping.test -t  i get some errors:
[image]
So i tried to run the Turtlebot Waffle demo and thats the output that i get:
[image]
Then Rviz launches:
[image]
And that's the output:
[image]

You can see that the map isn't creating

huangyoupingben, 2021-05-26:
I also have this problem，can anyone help?

huangyoupingben, 2021-05-26:
Hi, @fus3 ,I have  encountered the same problem.Have you solved the above problem?

maximilianwulf, 2021-05-31:
Hey @fus3 you don't see the elevation map because the pose publisher is crashing for you. This might be related to the turtlebot demo, which we cannot maintain here. Maybe you can ask the maintainers of the package.

agcxgz321, 2021-06-08:
I have encountered the same problem. Have you solved the problem?

maximilianwulf, 2021-06-08:
Is the waffle demo outputting a pose in your environment?

huangyoupingben, 2021-06-09:
yes,it is the waffle demo. I have not do any changes.

huangyoupingben, 2021-06-09:
There is no elevation map in the rviz when I run the waffle demo. The terminal show that  "Could not get pose information from robot for time 421.037000. Buffer empty?"

[image]

maximilianwulf, 2021-06-30:
Hey, I couldn't reproduce it on `noetic`: 

[image]

Are you using `melodic`?

maximilianwulf, 2021-06-30:
Can you try the `melodic-devel` branch? We just merged a fix.

huangyoupingben, 2021-07-04:
I fixed the problem,  the topic type of odometry published by waffle robot is differerent from the type in the code. I changed the code so that it can match the topic of odometry published by waffle robot. My ubuntu is 18.04.

maximilianwulf, 2021-07-06:
Hey @huangyoupingben, could you make a fixing MR for the `melodic-devel` branch? I think the community would appreciate it quite a bit.

wxk514, 2021-07-31:
@huangyoupingben Can you tell me how to modify it, thank you so much!!!

huangyoupingben, 2021-08-01:
@wxk514 ,focus on the type of " robotPoseCache_" in ElevationMapping.hpp.  You can use some tools ,for example rqt, to show the topic of odometry of waffle robot, and check the type of Odometry. In my case, they were different. I changed the type of "robotPoseCache_"  so that it can match the topic of odometry published by waffle robot, moreover, I changed some code related to "robotPoseCache_" in the function of ElevationMapping::pointCloudCallback which is in the ElevationMapping.cpp. you can have a try, good luch.

huangyoupingben, 2021-08-01:
@wxk514 ,focus on the type of " robotPoseCache_" in ElevationMapping.hpp. You can use some tools ,for example rqt, to show the topic of odometry of waffle robot, and check the type of Odometry. In my case, they were different. I changed the type of "robotPoseCache_" so that it can match the topic of odometry published by waffle robot, moreover, I changed some code related to "robotPoseCache_" in the function of ElevationMapping::pointCloudCallback which is in the ElevationMapping.cpp. you can have a try, good luck.

huangyoupingben, 2021-09-04:
@maximilianwulf ,  I have a question about your paper **The probabilistic Terrain Mapping for Mobile Robots with uncertain localiztion**. In the paper, the definition of the M frame is aligned with the z axis of the I frame, yaw(I,M)=yaw(I,B), in my opinion, map frame moves as the moving of robots relative to the fixed world frame. 
However, in the source code , it is found that the elevation map is based on a fixed world coordinate frame (fixed odometer coordinate frame) and will not change with the moving of the robot. When running elevation_map code, It is observed that the length and width of the output elevation map is always parallel to the fixed world coordinate frame. 
According to my understanding ,the map frame in your paper is different from the source code of elevaton_mapping.

huangyoupingben, 2021-09-04:
Dear @maximilianwulf ,I have a question about your paper **The probabilistic Terrain Mapping for Mobile Robots with uncertain localiztion.** In the paper, the definition of the M frame is aligned with the z axis of the I frame, yaw(I,M)=yaw(I,B). So,  the map frame moves as the moving of robots relative to the fixed world frame.
However, in the source code , it is found that the elevation map is based on a fixed world coordinate frame (fixed odometer coordinate frame) and will not change with the moving of the robot. When running elevation_map code, It is observed that the length and width of the output elevation map is always parallel to the fixed world coordinate frame.
**In summary the map frame in your paper is different from the source code of elevaton_mapping.**
look forward to your reply.

huangyoupingben, 2021-09-04:
Dear @maximilianwulf ,I have a question about your paper **The probabilistic Terrain Mapping for Mobile Robots with uncertain localiztion**. In the paper, the definition of the M frame is aligned with the z axis of the I frame, yaw(I,M)=yaw(I,B). So, **the map frame moves as the moving of robots relative to the fixed world frame.**
However, in the source code , it is found that the elevation map is based on a fixed world coordinate frame (fixed odometer coordinate frame) , and **the map frame will not change with the moving of the robot**. When running elevation_map code, It is observed that the length and width of the output elevation map is always parallel to the fixed world coordinate frame.
**In summary, the map frame in your paper is different from the source code of elevaton_mapping.**
look forward to your reply.

huangyoupingben, 2021-09-04:
Dear @maximilianwulf ,I have a question about your paper **The probabilistic Terrain Mapping for Mobile Robots with uncertain localiztion**. My question is that the map frame in your paper is different from the one in the source code of elevaton_mapping.
In your paper, the definition of the M frame is aligned with the z axis of the I frame, yaw(I,M)=yaw(I,B). So, the map frame moves as the moving of robots relative to the fixed world frame.
However, in the source code , it is found that the elevation map is based on a fixed world coordinate frame (fixed odometer coordinate frame) , and the map frame will not change with the moving of the robot. When running elevation_map code, It is observed that the length and width of the output elevation map is always parallel to the fixed world coordinate frame.
In summary, the map frame in your paper is different from the one in the source code of elevaton_mapping.
look forward to your reply.

huangyoupingben, 2021-09-04:
Dear @maximilianwulf ,I have a question about your paper **The probabilistic Terrain Mapping for Mobile Robots with uncertain localiztion.** My question is that **the map frame in your paper is different from the one in the source code of elevaton_mapping.**
In your paper, the definition of the M frame is aligned with the z axis of the I frame, yaw(I,M)=yaw(I,B). So, the map frame moves as the moving of robots relative to the fixed world frame.
However, in the source code , it is found that the elevation map is based on a fixed world coordinate frame (fixed odometer coordinate frame) , and the map frame will not change with the moving of the robot. When running elevation_map code, It is observed that the length and width of the output elevation map is always parallel to the fixed world coordinate frame.
In summary, the map frame in your paper is different from the one in the source code of elevaton_mapping.
look forward to your reply.

maximilianwulf, 2021-10-21:
> In summary, the map frame in your paper is different from the one in the source code of elevaton_mapping.
look forward to your reply.

You are free to choose the frame of reference depending on your needs. Do you have an issue?

zhanghua7099, 2021-11-28:
Same problem. Have anyone fixed this bug?