ANYbotics/elevation_mapping #161: fatal error: filters/filter_chain.hpp: No such file or directory

dlee640, 2021-05-08:
I get this error, and it seems like filters/filter_chain.hpp is not included in the repo. Is this from some other dependency or can you please provide this file?

Upon checking the PostprocessingPipelineFunctor.hpp, I realized that this file was updated recently by Magnus Gartner in Sep.14 2020.

```
slambox@slambox-NUC10i7FNH:~/elevation_ws$ catkin_make
Base path: /home/slambox/elevation_ws
Source space: /home/slambox/elevation_ws/src
Build space: /home/slambox/elevation_ws/build
Devel space: /home/slambox/elevation_ws/devel
Install space: /home/slambox/elevation_ws/install
Creating symlink "/home/slambox/elevation_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/slambox/elevation_ws/src -DCATKIN_DEVEL_PREFIX=/home/slambox/elevation_ws/devel -DCMAKE_INSTALL_PREFIX=/home/slambox/elevation_ws/install -G Unix Makefiles" in "/home/slambox/elevation_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/slambox/elevation_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/slambox/cartographer_ws/devel_isolated/cartographer_rviz;/home/slambox/cartographer_ws/install_isolated;/home/slambox/ouster_ws/devel;/home/slambox/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/slambox/cartographer_ws/devel_isolated/cartographer_rviz;/home/slambox/cartographer_ws/install_isolated;/home/slambox/ouster_ws/devel;/home/slambox/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/slambox/elevation_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - kindr
-- ~~  - kindr_msgs
-- ~~  - kindr_ros
-- ~~  - elevation_mapping
-- ~~  - elevation_mapping_demos
-- ~~  - kindr_rviz_plugins
-- ~~  - multi_dof_joint_trajectory_rviz_plugins
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'kindr'
-- ==> add_subdirectory(kindr)
-- Setting pure CMake build to false as not specified.
-- Building kindr as a catkin package.
-- Setting build type to 'Release' as none was specified.
[... 267 lines cut ...]
/home/slambox/elevation_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/postprocessing/PostprocessingPipelineFunctor.hpp:12:10: fatal error: filters/filter_chain.hpp: No such file or directory
 #include <filters/filter_chain.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/build.make:86: recipe for target 'elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/src/ElevationMap.cpp.o' failed
make[2]: *** [elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/src/ElevationMap.cpp.o] Error 1
In file included from /home/slambox/elevation_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/postprocessing/PostprocessorPool.hpp:20:0,
                 from /home/slambox/elevation_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/ElevationMap.hpp:29,
                 from /home/slambox/elevation_ws/src/elevation_mapping/elevation_mapping/src/ElevationMapping.cpp:24:
/home/slambox/elevation_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/postprocessing/PostprocessingPipelineFunctor.hpp:12:10: fatal error: filters/filter_chain.hpp: No such file or directory
 #include <filters/filter_chain.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/build.make:62: recipe for target 'elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/src/ElevationMapping.cpp.o' failed
make[2]: *** [elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/src/ElevationMapping.cpp.o] Error 1
[ 83%] Linking CXX shared library /home/slambox/elevation_ws/devel/lib/libkindr_rviz_plugins.so
[ 83%] Built target kindr_rviz_plugins
[ 86%] Linking CXX shared library /home/slambox/elevation_ws/devel/lib/libmulti_dof_joint_trajectory_rviz_plugins.so
[ 86%] Built target multi_dof_joint_trajectory_rviz_plugins
CMakeFiles/Makefile2:4379: recipe for target 'elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/all' failed
make[1]: *** [elevation_mapping/elevation_mapping/CMakeFiles/elevation_mapping_library.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```

yunxdai, 2021-05-17:
I meet the same problem when compiling on ros-melodic with Ubuntu18.04. I remembered this problem didn't exist when I used this package last time.

kosuke55, 2021-05-17:
How about switching `filter_chain.hpp` to `filter_chain.h`?

yunxdai, 2021-05-17:
> How about switching `filter_chain.hpp` to `filter_chain.h`?

Thank you very much for the suggestion! It works.

dlee640, 2021-05-19:
Issue solved by changing it to filter_chain.h
Thank you

closed 2021-05-19 by dlee640

maximilianwulf, 2021-05-31:
Please check out: https://github.com/ANYbotics/grid_map/pull/298

The issue comes from a mismatch between noetic and melodic. Master currently is targeting noetic, we created a melodic-devel branch to account for the changes.