ANYbotics/elevation_mapping #167: rviz not displaying anything, TF_OLD_DATA ignoring data from the past issue

dlee640, 2021-06-04:
Hello. 

I have collected actual data from quadruped (Unitree A1) & realsense D435i to test out elevation_mapping.

First of all, here is my bagfile info:
```
slambox@slambox-NUC10i7FNH:~$ rosbag info terrain.bag 
path:        terrain.bag
version:     2.0
duration:    39.6s
start:       Jun 03 2021 17:29:56.23 (1622755796.23)
end:         Jun 03 2021 17:30:35.80 (1622755835.80)
size:        1.5 GB
messages:    4700
compression: none [782/782 chunks]
types:       geometry_msgs/PoseWithCovarianceStamped [953b798c0f514ff060a53a3498ce6246]
             sensor_msgs/PointCloud2                 [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage                      [94810edda583a504dfda3829e70d7eec]
topics:      /camera/depth/color/points    781 msgs    : sensor_msgs/PointCloud2                
             /pose                         780 msgs    : geometry_msgs/PoseWithCovarianceStamped
             /tf                          3138 msgs    : tf2_msgs/TFMessage                      (4 connections)
             /tf_static                      1 msg     : tf2_msgs/TFMessage

```
As you can see, I have pointcloud topic, tf topic, and a pose topic that are in correct ros msg format.

Here is my modified `realsense_demo.launch' file. Two major changes here, adding the `<param name="/use_sime_time" value="true"/>` since I am playing the data from the bagfile, and adding the  `<node pkg="tf" type="static_transform_publisher" name="base2camera" args="0 0 0 0 0 0 1  /base_link /camera_link  20"/>`, since the transform between camera_link and the robot base_link was missing in my original bag file /tf...:
```
<launch>
  <param name="/use_sime_time" value="true"/>

  <!-- Elevation mapping node -->
  <node pkg="elevation_mapping" type="elevation_mapping" name="elevation_mapping" output="screen">
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/robots/simple_demo_robot.yaml" />
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/remove_object.yaml" />
    <rosparam command="load" file="$(find elevation_mapping)/config/sensor_processors/realsense_d435.yaml" />
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/postprocessing/postprocessor_pipeline.yaml" />
  </node>

  <!-- Launch visualizations for the resulting elevation map -->
  <include file="$(find elevation_mapping_demos)/launch/visualization.launch" />

  <!-- Launch RViz with the demo configuration -->
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find elevation_mapping_demos)/rviz/elevation_map_visualization_pointcloud.rviz" />

  <node pkg="tf" type="static_transform_publisher" name="base2camera" args="0 0 0 0 0 0 1  /base_link /camera_link  20"/>

</launch>
```

Here is my tf tree diagram for your information:
[image]

Here is my modified `simple_demo_robot.yaml`, and `remove_object.yaml`. I did not need to change realsense config file and the postprocessing file:

--> simple_demo_robot.yaml
```
point_cloud_topic: "/camera/depth/color/points"
map_frame_id: "map"
robot_base_frame_id: "base_link"
robot_pose_with_covariance_topic: "/pose"
robot_pose_cache_size: 200
track_point_frame_id: "base_link"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
```
--> remove_object.yaml
```
point_cloud_topic: "/camera/depth/color/points"
[... 121 lines cut ...]
[image]

When I try to play the bagfile again since my bagfile is kind of short, this error starts printing on the roslaunch terminal (terminal 2):
```
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame base_link at time 1.62276e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame base_link at time 1.62276e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.62276e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.62276e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame laser at time 1.62276e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame laser at time 1.62276e+09 according to authority unknown_publisher
```

I am not sure why nothing is showing on my rviz despite following all the procedures. If someone can spot my error, that would be appreciated. Thank you!

Miracle-qi, 2021-06-14:
I have similar problems

maximilianwulf, 2021-06-30:
Hey, I think your issue is here: 

```
[ WARN] [1622840350.875219243]: Could not load the input sources configuration from parameter
 /elevation_mapping/input_sources, are you sure it was pushed to the parameter server? Assuming
 that you meant to leave it empty. Not subscribing to any inputs!
 ```
 
 You should extend the configuration with: 
 
 ```
   input_sources:
    input_front:
      type: pointcloud
      topic: /camera/depth/color/points
      queue_size: 1
      publish_on_update: true
      sensor_processor:
        ignore_points_above: .inf
        ignore_points_below: -.inf
        type: structured_light
        cutoff_min_depth: 0.1
        cutoff_max_depth: 1.5
        normal_factor_a: 0.001063
        normal_factor_b:  0.003949
        normal_factor_c: 0.0
        normal_factor_d: 0.0008278
        normal_factor_e: 1
        lateral_factor: 0.01576 # approx 1 deg
 ```

Zhangwenyao1, 2021-12-14:
I have same question. And I want to know how to add the input_sources file and which type is it？