ANYbotics/elevation_mapping #169: elevation map sometimes normal and sometimes abnormal

wzy012, 2021-07-04:
Hi，
Thanks for your contribution!
I have some problems running this package:
Firstly,I use the odometry (nav_msgs/Odometry) type data to generate pose ,and point cloud to elevation map.
But it is have problems:
on terminal ,it is showing:Couldn't get pose information from robot for time…Buffer empty?
(rviz don't have elevation map)
But after a while，rviz has elevation map and terminal is not showing can'tget pose information….
After a while，elevation map is not update and terminal shows can't get pose information.
Can you get me some help or suggest ion.I don't know this reason.
Thanks!

maximilianwulf, 2021-07-06:
Hey @wzy012, could it be that you have to update the config file with the topic name?

wzy012, 2021-07-06:
> Hey @wzy012, could it be that you have to update the config file with the topic name? 

yes，I update the config file.

maximilianwulf, 2021-07-06:
Can you double-check the topic message type?

wzy012, 2021-07-06:
> Can you double-check the topic message type? 

year，I know！ Then I will reply to you after check.

closed 2021-07-13 by wzy012

maximilianwulf, 2021-07-16:
Hey, could you comment what helped you in the end?

wzy012, 2021-07-20:
The main cause of this problem is that pointcloud source.The package I used was changed by some company and it's time precise is low,low matching degree with the ros timestamp.The problem was solved after I switched to the official package.

maximilianwulf, 2021-07-21:
Great to hear :+1: