ANYbotics/elevation_mapping #172: waffle demo odom issue

christinad5, 2021-07-10:
Hi, I just installed this package on Ubuntu 20.04. I ran into a "Unknown frame odom" error when launching the waffle demo, and I haven't been able to solve the issue. Would love some help! Thank you! 
[image]

maximilianwulf, 2021-07-15:
Hey, the demo works on my machine. Can you show the output in the console?

christinad5, 2021-07-15:
Hi! Thanks so much for responding. I've attached images of my console output.
[image]
[image]
[image]

maximilianwulf, 2021-07-16:
You have Noetic, right?

maximilianwulf, 2021-07-16:
The output is the same on my machine. So it should work, or at least nothing is wrong with the console output.

MayHappybb, 2021-10-13:
try this:
$ export LC_NUMERIC="en_US.UTF-8"
then reboot the computer.

qiayuanliao, 2022-07-29:
I got the same issue. @christinad5 how can you solve this problem finally?

christinad5, 2022-08-01:
Unfortunately, I wasn't able to solve the issue

Hello970, 2023-04-05:
These days I meet the same problem with Ubuntu20.04 & ROS Noetic. Have you solved this problem?