ANYbotics/elevation_mapping #175: receiving pointclouds from realsense d400

rolandgvc, 2021-07-19:
Hi! I'm a bit confused about visualizing the elevation maps with the pointcloud generated by my realsense camera (d455). Here is my current configuration:

**realsense_demo.launch**
```
<launch>

  <!-- Elevation mapping node -->
  <node pkg="elevation_mapping" type="elevation_mapping" name="elevation_mapping" output="screen">
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/robots/d455.yaml" />
        <rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/simple_demo_map.yaml" />
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/postprocessing/postprocessor_pipeline.yaml" />
  </node>

  <!-- Launch visualizations for the resulting elevation map -->
  <include file="$(find elevation_mapping_demos)/launch/visualization.launch" />

  <!-- Launch RViz with the demo configuration -->
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find elevation_mapping_demos)/rviz/elevation_map_visualization_pointcloud.rviz" />

</launch>
```

**d455.yaml**
```
input_sources:
  camera:
    type: pointcloud
    topic: "/camera/depth_registered/points"
    queue_size: 1
    publish_on_update: true
    sensor_processor: 
      ignore_points_above: .inf
      ignore_points_below: -.inf
      cutoff_min_depth: 0.1
      cutoff_max_depth: 1.5
      type: structured_light # https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8768489
      normal_factor_a: 0.001063
      normal_factor_b:  0.003949
      normal_factor_c: 0.0
      normal_factor_d: 0.0008278
      normal_factor_e: 1
      lateral_factor: 0.01576 # approx 1 deg

map_frame_id: "odom"
robot_base_frame_id: "camera_link"
robot_pose_with_covariance_topic: "/pose"
robot_pose_cache_size: 200
track_point_frame_id: "base"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
```

**rviz**
[image]

**rqt_graph**
[image]

At the launch of `elevation_mapping` node, I'm getting these errors as well:
[image]

Please let me know if I'm doing something wrong or the next steps for visualizing the maps. Thanks!

maximilianwulf, 2021-07-19:
Can you try to shuffle from 

```
<rosparam command="load" file="$(find elevation_mapping_demos)/config/robots/d455.yaml" />
<rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/simple_demo_map.yaml" />
<rosparam command="load" file="$(find elevation_mapping_demos)/config/postprocessing/postprocessor_pipeline.yaml" />
```

to 

```
<rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/simple_demo_map.yaml" />
<rosparam command="load" file="$(find elevation_mapping_demos)/config/robots/d455.yaml" />
<rosparam command="load" file="$(find elevation_mapping_demos)/config/postprocessing/postprocessor_pipeline.yaml" />
```

Also can you give the output of `rosparam get /elevation_mapping` ?

rolandgvc, 2021-07-20:
Thanks for the tip! now `elevation_mapping` is starting, but giving the following error:
[image]

`rosparam get /elevation_mapping`
[image]

What do you think?

rolandgvc, 2021-07-20:
Nevermind, all working great!

maximilianwulf, 2021-07-21:
Great to hear. Can you say what the error in the end was?

closed 2021-07-21 by maximilianwulf

rolandgvc, 2021-07-21:
Running the camera config after the mapping config did it.

cc1416619381, 2021-10-10:
Hi, I'm working this package with realsense d435i, but a bit confused about the launch file and the corresponding yaml . I notice that your realsense_demo.launch is different from the default one, specifically, lacking of the following line:

<rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/remove_object.yaml" />

So, what's this param's function? Will it work normally after it is removed？
 
Thanks!

cc1416619381, 2021-10-10:
Hi, I'm working this package with realsense d435i, but a bit confused about the launch file and the corresponding yaml . I notice that your realsense_demo.launch is different from the default one, specifically, lacking of the following line:
`<rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/remove_object.yaml" />`
So, what's this param's function? Will it work normally after it is removed？

Thanks!

zhanghua7099, 2021-11-28:
Hi! Can you provide more details? I found the same errors.

Thanks.

wxk514, 2021-11-30:
> Running the camera config after the mapping config did it.

I met the same problem. could you provide more details? thanks!