ANYbotics/elevation_mapping #180: Could not get pose information from robot

wxk514, 2021-07-30:
When I tried to run the robotnik simulation, I got the following error:
[WARN] [1627651722.334626559, 1823.323000000]: Elevation map is updated without data from the sensor.
[ERROR] [1627651722.334710253, 1823.323000000]: Could not get pose information from robot for time 0.000000. Buffer empty?
[ERROR] [1627651722.334739088, 1823.323000000]: Updating process noise failed.
I have posted this topic: rosrun elevation_mapping_demos ground_tth_pose_publisher.py
How can I solve it? thank you very much

maximilianwulf, 2021-08-02:
Hey @wxk514, it seems your configuration of elevation mapping is wrong. Make sure to subscribe to the proper topics.

MatteoDeBenedetti, 2021-08-05:
Hi,  

I think I may have had the same problem, I digged a bit, and found something: there seem to be some mistakes in the new parameters' names and the corresponding documentation.

Here is my fix:

Pass both of these for the robot frame, because SensorProcessorBase uses the old one with robot_ in front:
```
base_frame_id: <base_link>
robot_base_frame_id: <base_link>
```

Also, the correct param to pass for the pose topic seems to be this one, instead of just robot_pose_topic as in the doc.
```
robot_pose_with_covariance_topic: <pose_topic>
```
Now for me it works fine.

@maximilianwulf I will keep looking around in the next days to prepare a pull request.
Thanks for the great library!

maximilianwulf, 2021-08-06:
@MatteoDeBenedetti submitting fixes directly is always the best :+1:

closed 2021-11-29 by wxk514