ANYbotics/elevation_mapping #181: Multi RealSense support

rolandgvc, 2021-08-03:
Is there support for mapping with multiple realsenses through the realsense_ros package?

maximilianwulf, 2021-08-03:
Yes, you can define as many input sources as you want.

rolandgvc, 2021-08-06:
I cannot get the pointclouds of multiple cameras, only the last one stated in `input_sources`. For example:
```
input_sources:
  camera:
    type: pointcloud
    topic: "/cam_2/depth/color/points"
    queue_size: 2
    publish_on_update: true
    sensor_processor: 
      #ignore_points_above: 1
      #ignore_points_below: -5
      cutoff_min_depth: 0.1
      cutoff_max_depth: 3.0
      type: structured_light # https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8768489
      normal_factor_a: 0.001063
      normal_factor_b:  0.003949
      normal_factor_c: 0.0
      normal_factor_d: 0.0008278
      normal_factor_e: 1
      lateral_factor: 0.11576 # approx 1 deg

    type: pointcloud
    topic: "/cam_3/depth/color/points"
    queue_size: 2
    publish_on_update: true
    sensor_processor: 
      #ignore_points_above: 1
      #ignore_points_below: -5
      cutoff_min_depth: 0.1
      cutoff_max_depth: 3.0
      type: structured_light # https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8768489
      normal_factor_a: 0.001063
      normal_factor_b:  0.003949
      normal_factor_c: 0.0
      normal_factor_d: 0.0008278
      normal_factor_e: 1
      lateral_factor: 0.11576 # approx 1 deg

map_frame_id: "odom"
robot_base_frame_id: "camera_link"
robot_pose_with_covariance_topic: "/pose"
robot_pose_cache_size: 200
track_point_frame_id: "camera_link"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
```
This will only show me the pointcloud from topic: `/cam_3/depth/color/points`

maximilianwulf, 2021-08-06:
Yes, because you are overwriting the first camera setup. Try: 

```
input_sources:
  camera_1:
    ...
  camera_2:
    ...
```

closed 2021-08-20 by rolandgvc