ANYbotics/elevation_mapping #182: No elevation map generated with VLP-16

zenified, 2021-08-24:
Hi, I am trying to experiment with using VLP-16 to generate the elevation map without any luck and I wonder if I could get some help.

The following are my configs
```                      
input_sources:                                                                  
  front:                                                                        
    type: pointcloud                                                            
    topic: /velodyne_points                                  
    queue_size: 1                                                               
    publish_on_update: true                                                     
    sensor_processor:                                                           
      type: laser                                                               
      #ignore_points_above: 0.0                                                 
      #ignore_points_below: -4.0                                                
      min_radius: 0.018                                                         
      beam_angle: 0.0006                                                        
      beam_constant: 0.0015                                                     
map_frame_id: "map"                                                             
robot_base_frame_id: "base_link"                                                
robot_pose_with_covariance_topic: "/converted_pose"                             
robot_pose_cache_size: 200                                                      
track_point_frame_id: "base_link"  
length_in_x: 5                                                                     
length_in_y: 5                                                                     
position_x: 0.0                                                                    
position_y: 0.0                                                                    
resolution: 0.08                                                                   
min_variance: 0.00009                                                              
max_variance: 0.01                                                                 
mahalanobis_distance_threshold: 2.5                                                
multi_height_noise: 0.00002
```
the default post processing pipeline config was used.

The debug message seems to indicate that pointcloud was received, so I am puzzled by the missing output. 

``` 
DEBUG] [1629796231.973022902, 1573567301.146515022]: Processing data from: base_lin
k                                                                                   
[DEBUG] [1629796231.974019094, 1573567301.146515022]: ElevationMap received a point 
cloud (20631 points) for elevation mapping.                                         
[DEBUG] [1629796231.974109494, 1573567301.146515022]: Sensor Processor processing fo
r frame base_link                                                                   
[DEBUG] [1629796231.977267360, 1573567301.146515022]: Limiting point cloud to the he
ight interval of [-5.000000, 0.000000] relative to the robot base.
[DEBUG] [1629796231.981007578, 1573567301.156571482]: removePointsOutsideLimits() re
duced point cloud to 12331 points.
[DEBUG] [1629796231.984386808, 1573567301.156571482]: Elevation map raw has been pub
lished.
[DEBUG] [1629796232.225100060, 1573567301.397873844]: Elevation map is checked for r
elocalization.
[DEBUG] [1629796232.225213325, 1573567301.397873844]: Elevation map has been moved t
o position (66.760000, 29.530000).
[DEBUG] [1629796232.225232731, 1573567301.397873844]: Updating map with latest predi
ction from time 1573567301.116196.
[DEBUG] [1629796232.267988432, 1573567301.438091970]: Raw map has been updated with 
```
I wonder if something is off in my configurations. Thanks in advance

AtsukiOsanai, 2021-09-08:
Please increase the queue_size more than 3.
It worked for me.

zenified, 2021-09-08:
I also realized, increasing the map size to larger then 10 helps

closed 2021-09-08 by zenified