ANYbotics/elevation_mapping #188: Loading map only resets map

Max-Ole, 2021-11-03:
Hello, 
I have saved an elevation map using the save_map service call. Now would like to load it back into the elevation mapping package using the load_map rosservice call.

I first run the same launch file which I also used to save the map. It brings up a simulation in gazebo and rviz to visualize the elevation map. However if I then call 
`rosservice call /elevation_mapping/load_map "file_path: '/home/ubuntu/Desktop/BA_MaxOle_ws/myRecordedData/myRecordedData/elevation_map_geb64_2.bag'`  (returns `success: True`)
then the map that was being created until then disappears and a new map shows up from the current sensor input. There is no sign of the map I loaded.
I tried to use the service call `disable_updates` because I thought the new input was overriding the loaded map. But with new input disabled the map disappears without replacement. 

Am I misusing the service call? What I am trying to achieve is to have the same map data to test different elevation_mapping postprocessor pipelines and the conversion to a occupancy grid using grid_map_visualization from the grid_map package.

Also please note that `topic_name` was omitted when saving and loading the map because of the following error in the call:
```
ubuntu@ubuntu:~/Desktop/BA_MaxOle_ws/src/my_work_pkg/launch$ rosservice call /elevation_mapping/save_map "file_path: '/home/administrator/catkin_ws_Max-OleVW/myRecordedData/elevation_map_geb64_3.bag' topic_name: 'my_topic_name'"
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/rosservice", line 35, in <module>
    rosservice.rosservicemain()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosservice/__init__.py", line 749, in rosservicemain
    _rosservice_cmd_call(argv)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosservice/__init__.py", line 610, in _rosservice_cmd_call
    service_args.append(yaml.safe_load(arg))
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 93, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block mapping
  in "<string>", line 1, column 1:
    file_path: '/home/administrator/ ... 
    ^
expected <block end>, but found '<scalar>'
  in "<string>", line 1, column 95:
     ... Data/elevation_map_geb64_3.bag' topic_name: 'my_topic_name'
```
The file was saved using `rosservice call /elevation_mapping/save_map "file_path: '/home/administrator/catkin_ws_Max-OleVW/myRecordedData/elevation_map_geb64_2.bag'"` and then moved from the robot to my PC.

Thanks for any advice!

maximilianwulf, 2021-11-05:
Can you see the elevation map by just replaying the bagfile, with `rosbag play /home/administrator/catkin_ws_Max-OleVW/myRecordedData/elevation_map_geb64_3.bag` ?

Max-Ole, 2021-11-08:
When I use `rosbag play` rviz doesn't show the elevtion map. I thought that was because the bagfile only contains a single message. 
I wrote a simple script that receives the map when I use the command you mentioned. It then repeatedly publishes that map on the same ros topic. By doing that the elevation map can be displayed in rviz.

maximilianwulf, 2021-11-09:
Thank you for testing.

> However if I then call
rosservice call /elevation_mapping/load_map "file_path: '/home/ubuntu/Desktop/BA_MaxOle_ws/myRecordedData/myRecordedData/elevation_map_geb64_2.bag' (returns success: True)
then the map that was being created until then disappears and a new map shows up from the current sensor input. There is no sign of the map I loaded.

Could it be that when replaying the robot is in another position? Based on the `(returns success: True)` the loading was successful. Saved maps have a position attached to it, depending on where your robot is at the point of replaying the map could be outside of the range.

Max-Ole, 2021-11-09:
You are correct, the loading is successful and it can be seen in the different output of `rostopic echo /elevation_mapping/elevation_map`. And the map shows up in rviz.

For anyone having a similar problem, here is what I had to do to find the map:
- Elevation_mapping updates need to be disabled, otherwise loaded data will be overwritten.
- The correct frame of the map needs to be present. In my case I launched gazebo.
- Color layer in rviz needs to be set to a layer that is present in the loaded file or else nothing will show up.
- The map is no longer robot centric and may be far away from robot.

Thanks for the help!

closed 2021-11-09 by Max-Ole

maximilianwulf, 2021-11-09:
Thank you for confirming that it worked.