ANYbotics/elevation_mapping #189: How to get static map frame

naoya-kumagai, 2021-11-11:
To the best of my understanding, the elevation map is overwritten with the height relative to the robot's pose. Therefore, when observing the same plane with the same height, the height will be different if the robot (camera) is at a different height. (The resulting map seems "patched together" from different observations")
How can I get a map with a consistent world frame? This would be very nice since the surface normal and elevation mapping features are very useful for this package.

maximilianwulf, 2021-11-11:
Hey @naoya-kumagai, 

> How can I get a map with a consistent world frame? This would be very nice since the surface normal and elevation mapping features are very useful for this package.

It does exactly this. If you check out this [config](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/config/robots/waffle_robot.yaml#L2) you see that the elevation map is described in the odom frame. In case your pose is not drifting you will get the result you want. 
In case your pose is actually drifting then you will see the mismatch and you will have to fix the drift in your pose.

naoya-kumagai, 2021-11-18:
I see, I might also be doing something wrong.
The packages to the pose and covariance of the robot, as described in the README.
Should this be the pose of the sensor that is getting the pointcloud?

maximilianwulf, 2021-11-22:
> Should this be the pose of the sensor that is getting the pointcloud?

Not necessarily, but it can be.