ANYbotics/elevation_mapping #190: Postprocessor Pipeline with SlidingWindow: no default buffer start index.

Max-Ole, 2021-11-14:
Hello, not sure whether to ask here or at the grid_map page.

I was using the post-processing feature and then tried to add a `SlidingWindowMathExpressionFilter`. This however results in this error as soon as the robot is moved or a map is loaded:
`Postprocessor pipeline, thread 0 experienced an error: SlidingWindowIterator cannot be used with grid maps that don't have a default buffer start index.`

Is there an option anywhere to set the default buffer start index? I am not interacting with the map or package programmatically, only through the launch file and referenced yaml files.

I also tried to use a DuplicationFilter in case the SlidingWindowMathExpressionFilter can't create  a new layer. This did not seem to change anything. Here is part of the yaml file for the postprocessor pipeline:
```
postprocessor_pipeline:
  # [ ... ]

  # Duplicate a layer to get a default buffer start index. (fix attempt)
   - name: duplicate_3
     type: gridMapFilters/DuplicationFilter
     params:
       input_layer: elevation
       output_layer: output

  # get height of edge
   - name: math_expression
     type: gridMapFilters/SlidingWindowMathExpressionFilter
     params:
       input_layer: elevation
       output_layer: output
       expression: meanOfFinites(elevation) # Box blur from example to test/debug this filter
       # expression: max(elevation) - min(elevation) # planned use: get height of edge as max height difference in region around edge
       compute_empty_cells: false
       edge_handling: crop # options: inside, crop, empty, mean
       window_size: 5 # in number of cells
```

maximilianwulf, 2021-11-15:
Hm interesting :thinking: so yes there is a function inside grid map that sets the buffer to the default start index. 

This seems to be a bug, could you try to add the function here: https://github.com/ANYbotics/grid_map/blob/539f9db782e73cd48a112a6a72f3dbc11868d80f/grid_map_filters/src/SlidingWindowMathExpressionFilter.cpp#L83 and see if it solves your issue?

Max-Ole, 2021-11-17:
Thank you for your help. I have solved the issue by adding `convertToDefaultStartIndex()` at the line you recommended in `SlidingWindowMathExpressionFilter.cpp`. It works well now.

The mapIn is copied because it is given as being constant and therefore the start index cannot be changed. Maybe the const can be ommited instead to make it run more efficiently.
```
bool SlidingWindowMathExpressionFilter<T>::update(const T& mapIn, T& mapOut)
{
  grid_map::GridMap mapInCopy = mapIn; // copy map since given mapIn is const so startIndex can't be changed
  mapInCopy.convertToDefaultStartIndex();
  mapOut = mapInCopy;

  mapOut.add(outputLayer_);
  Matrix& outputData = mapOut[outputLayer_];
  grid_map::SlidingWindowIterator iterator(mapInCopy, inputLayer_, edgeHandling_, windowSize_);
  if (useWindowLength_) iterator.setWindowLength(mapInCopy, windowLength_);
  // [...]
}
```

closed 2021-11-17 by Max-Ole

Magnusgaertner, 2021-12-15:
Addendum: 

You can also use the filter as is, but make sure the grid map has default start index by using the BufferNormalizerFilter which essentially does what you have edited in the Sliding math expression filter. Both ways work fine.

Max-Ole, 2021-12-15:
That is exactly what I need, thank you. I was unaware that there were other filters than those described in the grid_map README.md file but now I found the corresponding code in the grid_map_filters folder.