ANYbotics/elevation_mapping #191: Dynamic Update of the Elevation map

aled96, 2021-11-19:
I am using this elevation map for autonomous navigation. In order to be more robust against localization errors and moving/dynamic obstacles how can I let the new data from the sensor overwrite what we have already in the elevation map ?

In this way the new data will be always aligned to the current robot position and sensors output and the map will be update consequently.

Thanks !

maximilianwulf, 2021-11-22:
Hi @aled96, do you want to discard the history of each cell?

aled96, 2021-11-22:
Hi, yes but only for the cells in which I have new data from the last point cloud acquired. This can help with localization errors that would bring errors in the map.

Is this option available or I should write a node from scratch ?

maximilianwulf, 2021-11-22:
This is currently not an official use-case of elevation mapping, and therefore unfortunately not yet available.

aled96, 2021-11-23:
Ok thank you for the answer. I will close the issue.

closed 2021-11-23 by aled96