ANYbotics/elevation_mapping #192: Questions about the Realsense D455 Demo

zhanghua7099, 2021-11-28:
Hi! 

I encountered the same error as this closed [issue](https://github.com/ANYbotics/elevation_mapping/issues/175#issuecomment-883311092).

Could you please help me to fix this bug? The author seems to solve this problem. But I can't understand his [method](https://github.com/ANYbotics/elevation_mapping/issues/175#issuecomment-884046441).

maximilianwulf, 2021-11-29:
Hi @zhanghua7099 can you share your configuration?

zhanghua7099, 2021-12-01:
My configuration file is as the same as the closed issue. 

Ubuntu 20.04+Ros noetic

realsense_demo.launch
```
<launch>

  <!-- Elevation mapping node -->
  <node pkg="elevation_mapping" type="elevation_mapping" name="elevation_mapping" output="screen">
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/simple_demo_map.yaml" />
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/robots/d455.yaml" />
    <rosparam command="load" file="$(find elevation_mapping_demos)/config/postprocessing/postprocessor_pipeline.yaml" />  
  </node>

  <!-- Launch visualizations for the resulting elevation map -->
  <include file="$(find elevation_mapping_demos)/launch/visualization.launch" />

  <!-- Launch RViz with the demo configuration -->
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find elevation_mapping_demos)/rviz/elevation_map_visualization_pointcloud.rviz" />

</launch>
```
d455.yaml
```
input_sources:
  camera:
    type: pointcloud
    topic: "/camera/depth/color/points"
    queue_size: 1
    publish_on_update: true
    sensor_processor: 
      ignore_points_above: .inf
      ignore_points_below: -.inf
      cutoff_min_depth: 0.1
      cutoff_max_depth: 1.5
      type: structured_light # https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8768489
      normal_factor_a: 0.001063
      normal_factor_b:  0.003949
      normal_factor_c: 0.0
      normal_factor_d: 0.0008278
      normal_factor_e: 1
      lateral_factor: 0.01576 # approx 1 deg

map_frame_id: "odom"
robot_base_frame_id: "camera_link"
robot_pose_with_covariance_topic: "/pose"
robot_pose_cache_size: 200
track_point_frame_id: "base"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
```
rqt_graph
[image]

elevation mapping console
[image]

maximilianwulf, 2021-12-01:
Hey, given the console output I don't see an issue. Can you say what issue you experience? 

Do you have a pose source?

zhanghua7099, 2021-12-07:
Hi! 

I use the rtab-map as the pose source. The parameter `robot_pose_with_covariance_topic` in `d455.yaml` was changed to:
`robot_pose_with_covariance_topic: "/rtabmap/odom"`

[image]

How to solve this problem?

THX!

zhanghua7099, 2021-12-07:
Fixed by change VO estimater.

closed 2021-12-07 by zhanghua7099