ANYbotics/elevation_mapping #193: Why are there so many holes and how to solve them?

DLonng, 2021-11-29:
[image]

maximilianwulf, 2021-11-29:
Hey @DLonng, it seems your point clouds are not dense enough, 2 possible options: 

- When the robot moves the holes get filled once a point from a point cloud falls into it. 
- You add a postprocessor pipeline to the elevation mapping which fills small holes, we currently recommend this one: https://github.com/ANYbotics/grid_map/blob/master/grid_map_filters/include/grid_map_filters/MedianFillFilter.hpp

DLonng, 2021-11-30:
> 
[image]
very good!!! hahhaaa :)

maximilianwulf, 2021-11-30:
Nice, seems to work. Will close the issue, feel free to re-open if you need further assistance.

closed 2021-11-30 by maximilianwulf

hzyhuter, 2022-10-29:
hey，can you teach me how to use the MedianFillFilter? thank you very much!