ANYbotics/elevation_mapping #195: D435i problem

wxk514, 2021-11-29:
Hi, An error occurred when I tried to run realsense_demo.launch with D435i.
[image]
[image]

The content of realsense_demo.launch and realsense_d435.yaml is:
[image]
[image]
[image]

could you help me? thanks!

wxk514, 2021-11-30:
When I first run my realsense terminal and then run realsense demo, the following error occurs
`moro@moro-Lenovo-WEI6-14IKB:~/elevationmap_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/moro/.ros/log/7d3c4d52-5199-11ec-83f9-505bc2b64fe5/roslaunch-moro-Lenovo-WEI6-14IKB-22076.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://moro-Lenovo-WEI6-14IKB:36727/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 251 lines cut ...]
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find elevation_mapping_demos)/rviz/elevation_map_visualization_pointcloud.rviz" />

</launch>
`
the config is:
`input_sources:
  front:
    type: pointcloud
    topic: /camera/depth/color/points
    queue_size: 5
    publish_on_update: true
    sensor_processor:
      ignore_points_above: .inf
      ignore_points_below: -.inf
      type: structured_light
      cutoff_min_depth: 0.1
      cutoff_max_depth: 1.5
      normal_factor_a: 0.001063
      normal_factor_b:  0.003949
      normal_factor_c: 0.0
      normal_factor_d: 0.0008278
      normal_factor_e: 1
      lateral_factor: 0.01576 # approx 1 deg
`
It is ok for me to run the turtlebot in the simulation environment. could you help me ,thanks!

wxk514, 2021-11-30:
I modify my realsense_d435.yaml as follows:
[image]
Then I turned on my camera terminal: roslaunch realsense_camera rs_camera.launch, then I run realsense_demo.launch file
The following error occurred:
`[ WARN] [1638248063.325983165]: Could not find the parameter: `algorithm`. Setting to default value: 'area'.
[ WARN] [1638248063.326951051]: Could not find the parameter: `parallelization_enabled`. Setting to default value: 'false'.
[ WARN] [1638248063.326972763]: Could not find the parameter: `thread_number`. Setting to default value: 'automatic'.
[ INFO] [1638248063.328021982]: Elevation mapping node started.
[ INFO] [1638248063.344738281]: Elevation map grid resized to 600 rows and 600 columns.
[ WARN] [1638248063.371231195]: Parameter 'point_cloud_topic' is deprecated, please use 'input_sources' instead.
[ INFO] [1638248063.373057863]: Subscribing to pointcloud: /camera/depth/color/points, queue_size: 5.
[ INFO] [1638248063.378049165]: Elevation mapping node initializing ... 
[ INFO] [1638248064.378400529]: Done initializing.
[ INFO] [1638248064.379243554]: First corresponding point cloud and pose found, elevation mapping started. 
[ERROR] [1638248064.379374187]: Could not get pose information from robot for time 1638248064.000550. Buffer empty?
[ERROR] [1638248064.379687710]: Could not get pose information from robot for time 1638248064.067268. Buffer empty?
[ERROR] [1638248064.379755810]: Could not get pose information from robot for time 1638248064.133986. Buffer empty?
[ERROR] [1638248064.379806554]: Could not get pose information from robot for time 1638248064.200704. Buffer empty?
[ERROR] [1638248064.379864967]: Could not get pose information from robot for time 1638248064.267423. Buffer empty?
[ERROR] [1638248064.427233473]: Could not get pose information from robot for time 1638248064.334141. Buffer empty?
[ERROR] [1638248064.502884703]: Could not get pose information from robot for time 1638248064.400859. Buffer empty?
[ERROR] [1638248064.561115293]: Could not get pose information from robot for time 1638248064.467577. Buffer empty?
[ERROR] [1638248064.628894612]: Could not get pose information from robot for time 1638248064.534295. Buffer empty?
[ERROR] [1638248064.693376261]: Could not get pose information from robot for time 1638248064.601013. Buffer empty?
[ERROR] [1638248064.766576165]: Could not get pose information from robot for time 1638248064.667732. Buffer empty?
[ERROR] [1638248064.828393895]: Could not get pose information from robot for time 1638248064.734450. Buffer empty?
[ERROR] [1638248064.895281299]: Could not get pose information from robot for time 1638248064.801169. Buffer empty?
[ERROR] [1638248064.962047742]: Could not get pose information from robot for time 1638248064.867888. Buffer empty?
[ERROR] [1638248065.028726487]: Could not get pose information from robot for time 1638248064.934607. Buffer empty?
[ERROR] [1638248065.094361082]: Could not get pose information from robot for time 1638248065.001326. Buffer empty?`

maximilianwulf, 2021-11-30:
Hey, the current problem is

```
Could not get pose information from robot for time 1638248064.334141. Buffer empty? [ERROR]
```

Elevation mapping requires a pose input as well, on the real robot this normally comes from a state estiamator, but this example does not start an extra one and assumes it already exists somewhere. The waffle bot demo also provides a pose.

wxk514, 2021-11-30:
Thank you, but I don’t know how to modify it. Can you provide a detailed example about the pose？Thanks so much!

maximilianwulf, 2021-11-30:
> Can you provide a detailed example about the pose？

You need a state estimator that provides a pose of the robot. This usually comes from other packages, check this one out for example: https://github.com/ori-drs/pronto

wxk514, 2021-12-01:
Thank you. I'll try