ANYbotics/elevation_mapping #196: Example for Multiple Sensors?

cyoahs, 2021-12-07:
Hello.

Currently we can use multiple cameras in input_resources, but how can we specify the relative position from each camera to the robot base? Can you provide a more specific example for this usage? 

Thanks!

cyoahs, 2021-12-07:
I have understand this. I will reopen this issue with a launch file and a config file for anyone needed in the future.

closed 2021-12-07 by cyoahs

maximilianwulf, 2021-12-07:
For future travellers, 

> but how can we specify the relative position from each camera to the robot base? Can you provide a more specific example for this usage?

with robot urdf file, or general `tf2` information. 

> Can you provide a more specific example for this usage?

feel free to submit an example.