ANYbotics/elevation_mapping #197: Migration to ROS2

KirillMouraviev, 2021-12-10:
Hi. Are there any plans to port this project on ROS2?

maximilianwulf, 2021-12-10:
Hey @KirillMouraviev, not yet official plans. But it can happen anytime. 

For https://github.com/ANYbotics/grid_map the open source community works on a ROS2 version, you can check it on the ros2 branch. 

If there is someone willing to start a port, we can already support it.

remod, 2022-02-18:
[Done](https://github.com/ANYbotics/elevation_mapping/tree/ros2)

SivertHavso, 2022-02-19:
Looking at https://github.com/ANYbotics/grid_map/pull/245, most of the merge conflicts seem to be from code formatting changes. Things like the include order, keeping lines <100 characters, the initialization order in classes, using the c++ style `static_cast<type>(value)` instead of `(type)value` casts, etc.. I see two options, either linting and applying these formatting changes to both branches (I would prefer this option), or not using the same code style as ROS2 packages in general use and disable linting as part of the test process.

I've also opened https://github.com/ANYbotics/elevation_mapping/pull/203 to move from tf to tf2 on the ROS1 branch which would avoid some future merge conflicts.

maximilianwulf, 2022-05-03:
> I am unsure about the learnings made in the ROS2 transition of grid map. Is there anything to be careful of?

Yes, the biggest learning is the different formatting :thinking: So we are a bit stuck with porting the ROS1 changes. @SivertHavso if you want you can help us por the recent changes to ROS2.

ktk1501, 2022-05-11:
Hi, @SivertHavso

Do you have a plan to upload the ROS2 ported version? It will be greatly helpful for me even if the implementations are dirty.

samprt-heracles, 2022-09-30:
Hi @SivertHavso ! Any news on a first version of the ros2 implementation ?

NickNair, 2023-03-29:
@maximilianwulf @SivertHavso 

Does this project have a UML/ class documentation. This will certainly help with the ros2 port.

If anybody has a decently functional ros2 fork do share.

NickNair, 2023-03-30:
> @maximilianwulf I've got some extra time I could use on a ROS2 port. I've had a go at a quick-and-dirty port of it already (that we use internally), so I have an idea of the scope of the process. If you can create a ROS2 branch and you (or someone else) are willing to review the pull requests, I'm happy to get started.

Does this port have all the InputSouceManager, Sensor ProcessorBase classes ported to ROS2 ?

adyczech, 2023-08-07:
> If anybody has a decently functional ros2 fork do share.

If anybody is interested, here is a crude port of elevation_mapping to ROS2 (foxy) https://github.com/adyczech/elevation_mapping/tree/ros2. It's definitely not pull request ready and many features (services, timers, threading, testing) are still yet to be ported, but the main functionality is working.

I'll try to keep working on porting the rest of the features but it's not my main priority now. But feel free to report any issues and features, that are required.