ANYbotics/elevation_mapping #199: Waiting for tf transformation to be available. (Message is throttled, 10s.)

zhanghua7099, 2022-01-05:
Hi! 

I found that elevation mapping needs the transformation tree as input. For now, I didn't have this input. Can you give some details?

Thank you!

Magnusgaertner, 2022-01-26:
Hi @zhanghua7099, 

you may want to have a look at the `Simple Demo` for a static setup to use elevation_mapping. 
In general, you are right that elevation_mapping expects a transformation tree to be available. Thus, the simplest option is to add a static transform publisher to specify a rigid, static transform 

```yaml
  <node pkg="tf" type="static_transform_publisher" name="world_to_robot" args="2.0 6.0 0 0 0.0 0 /map /base 100"/>

```

See 
- https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/launch/simple_demo.launch#L19
- https://github.com/ANYbotics/elevation_mapping/blob/master/README.md#simple-demo--ground-truth-demo

zhanghua7099, 2022-02-09:
Thank you very much! The code runs well.

closed 2022-02-09 by zhanghua7099