ANYbotics/elevation_mapping #202: Generated elevation_map is floating in the air

Yaru-Gu, 2022-02-15:
Hello,

Thank you for sharing this remarkable work!
I'm trying to use this package with a quadruped robot. I tested using the environment _turtlebot3_house.world_ for my robot to walk through. Everything else works fine, but the generated elevation_map is not standing on the ground. Please see the attached image for better illustration. The mapped floor of the _turtlebot3_house.world_ is floating in the air, and my robot is buried by it. 
[image]
 
I have no clue where to adjust it. Any help would be very much appreciated.

zhanghua7099, 2022-02-24:
Hi! It may be caused by the initial coordinate system of the SLAM algorithm. I guess the initial pose is not aligned to gravity. 

You can try to adjust the orientation of the camera (parallel to the ground). Or try to align the initial coordinate system with the direction of gravity.

Yaru-Gu, 2022-03-10:
This problem has been solved by using more precise inertial values for all links.

closed 2022-03-10 by Yaru-Gu