ANYbotics/elevation_mapping #205: Elevation Map not Displaying in RViz

nbertschmann, 2022-03-28:
I have been attempting to use ZED point cloud data to test out the elevation_mapping package. 

First, here is the output I get using `rqt_graph`. The elevation_mapping node is subscribed to the down-sampled point cloud (type sensor_msgs/PointCloud2), the pose of the ZED camera (type geometry_msgs/PoseWithCovariance), and tf.

[image]

In RViz, the pointcloud from the ZED camera displays properly, but the elevation mapping data from elevation_mapping_raw is not displaying. Below is a screenshot of my RViz window:

[image]

My bagfile info is below. As you can see, the /elevation_mapping/elevation_map topic sent 31 messages. 

```
path:        2022-03-28-01-47-54.bag
version:     2.0
duration:    23.0s
start:       Mar 28 2022 01:48:04.00 (1648446484.00)
end:         Mar 28 2022 01:48:26.97 (1648446506.97)
size:        483.1 MB
messages:    1274
compression: none [70/70 chunks]
types:       bond/Status                             [eacc84bf5d65b6777d4c50f463dfb9c8]
             diagnostic_msgs/DiagnosticArray         [60810da900de1dd6ddd437c3503511da]
             dynamic_reconfigure/Config              [958f16a05573709014982821e6822580]
             dynamic_reconfigure/ConfigDescription   [757ce9d44ba8ddd801bb30bc456f946f]
             geometry_msgs/PoseStamped               [d3812c3cbc69362b77dc0b19b345f8f5]
             geometry_msgs/PoseWithCovarianceStamped [953b798c0f514ff060a53a3498ce6246]
             grid_map_msgs/GridMap                   [95681e052b1f73bf87b7eb984382b401]
             nav_msgs/Odometry                       [cd5e73d190d741a2f92e81eda573aca7]
             nav_msgs/Path                           [6227e2b7e9cce15051f669a5e197bbf7]
             rosgraph_msgs/Log                       [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/CameraInfo                  [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage             [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image                       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2                 [1158d486dd51d683ce2f1be655c3c181]
             stereo_msgs/DisparityImage              [04a177815f75271039fa21f16acad8c9]
             tf2_msgs/TFMessage                      [94810edda583a504dfda3829e70d7eec]
             theora_image_transport/Packet           [33ac4e14a7cff32e7e0d65f18bb410f3]
             visualization_msgs/Marker               [4048c9de2a16f4ae8e0538085ebf1b97]
topics:      /camera/depth/points_downsampled                                                   29 msgs    : sensor_msgs/PointCloud2                
             /diagnostics                                                                       19 msgs    : diagnostic_msgs/DiagnosticArray        
             /elevation_map_fused_visualization/elevation_cloud                                 26 msgs    : sensor_msgs/PointCloud2                
             /elevation_map_fused_visualization/lower_bound_cloud                               28 msgs    : sensor_msgs/PointCloud2                
             /elevation_map_fused_visualization/map_region                                      28 msgs    : visualization_msgs/Marker              
             /elevation_map_fused_visualization/upper_bound_cloud                               29 msgs    : sensor_msgs/PointCloud2                
             /elevation_mapping/elevation_map                                                   31 msgs    : grid_map_msgs/GridMap                  
             /elevation_mapping/visibility_cleanup_map                                          31 msgs    : grid_map_msgs/GridMap                  
             /pcl_manager/bond
...
```    

It seems as though the elevation mapping package is working correctly, so I am not sure why it won't display in RViz. 

Not sure if this helps, but here is the output from  `roslaunch elevation_mapping_demos zed_test2.launch`  (modified version of simple_demo to test elevation mapping with ZED input):
```
nbertsch@nbertsch-desktop:~/catkin_ws3$ roslaunch elevation_mapping_demos zed_test2.launch
... logging to /home/nbertsch/.ros/log/48b65074-ae59-11ec-a3cb-1c1bb5e39bda/roslaunch-nbertsch-desktop-26528.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nbertsch-desktop:34905/

SUMMARY
========

PARAMETERS
 * /elevation_map_fused_visualization/grid_map_topic: /elevation_mappin...
 * /elevation_map_fused_visualization/grid_map_visualizations: [{'type': 'point_...
 * /elevation_map_raw_visualization/grid_map_topic: /elevation_mappin...
[... 88 lines cut ...]
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1648445990.075174573]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446000.767268060]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446011.369331966]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446022.212904541]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446032.716968642]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446043.413005024]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446054.343865369]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446065.242892720]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446076.209670638]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446087.149059940]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446098.100398033]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446109.049564584]: Waiting for tf transformation to be available. (Momeone help me get theessage is throttled, 10s.)
[ INFO] [1648446120.023578234]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446130.995318400]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446141.142775713]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446151.238169288]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446161.239057705]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648446172.153405421]: Waiting for tf transformation to be available. (Message is throttled, 10s.)

```
I would be greatly appreciated if I could get some tips on how to display the elevation map properly!

maximilianwulf, 2022-03-30:
Hey, setup should look okay, 

I guess there might be problem with the tf/pose information, i.e. the following part from the console: 

```
[ERROR] [1648445958.063837999]: Could not get pose information from robot for time 1648445956.360284. Buffer empty?
[ INFO] [1648445958.736752215]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
```

nbertschmann, 2022-03-30:
@maximilianwulf Is there any way to subscribe to a "fake pose" to see if my pose input is the issue?

maximilianwulf, 2022-03-30:
Hm, yes, but you have to create the fake pose. 

I would actually try the waffle demo first and then just rewire the point clouds, and see if it reacts.

nbertschmann, 2022-03-30:
@maximilianwulf Thank you for the advice - I just tried this out.

I'm still getting the same issue where the `elevation_mapping/elevation_map` topic is being published, but `elevation_mapping/elevation_map_raw` is not.

Here is what I modified in the `turtlesim3_wafffle_demo.launch` file. The input to the down-sample filter is now `/zed/zed_node/point_cloud/cloud_registered`

```e
  <!-- Run a passthrough filter to down-sample the sensor point cloud.-->
  <node pkg="nodelet" type="nodelet" name="pcl_manager" args="manager" output="screen"/>
  <node pkg="nodelet" type="nodelet" name="voxel_grid" args="load pcl/VoxelGrid pcl_manager" output="screen">
    <remap from="~input" to="/zed/zed_node/point_cloud/cloud_registered"/> 
    <remap from="~output" to="/camera/depth/points_downsampled"/>
    <rosparam>
      filter_field_name: z
      filter_limit_min: 0.01
      filter_limit_max: 6.0  <!-- Limit the range of points to forward to 6 meters in viewing direction-->
      filter_limit_negative: False
      leaf_size: 0.05  <!-- Limit the density to 5 cm-->
    </rosparam>
  </node>
```
**rqt_graph:**

[image]

**RViz Window:**

[image]

**Terminal Output:**

```
nbertsch@nbertsch-desktop:~/catkin_ws$ roslaunch elevation_mapping_demos turtlesim3_waffle_demo2.launch
... logging to /home/nbertsch/.ros/log/0ea855d8-b054-11ec-a556-1c1bb5e39bda/roslaunch-nbertsch-desktop-10065.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://nbertsch-desktop:46879/

SUMMARY
========

PARAMETERS
 * /elevation_mapping/enable_visibility_cleanup: False
 * /elevation_mapping/fused_map_publishing_rate: 0.5
 * /elevation_mapping/init_submap_height_offset: 0.01
 * /elevation_mapping/initialization_method: 0
 * /elevation_mapping/initialize_elevation_map: False
 * /elevation_mapping/input_sources/zed/publish_on_update: True
 * /elevation_mapping/input_sources/zed/queue_size: 1
 * /elevation_mapping/input_sources/zed/sensor_processor/apply_voxelgrid_filter: True
 * /elevation_mapping/input_sources/zed/sensor_processor/cutoff_max_depth: 20
 * /elevation_mapping/input_sources/zed/sensor_processor/cutoff_min_depth: 0.3
 * /elevation_mapping/input_sources/zed/sensor_processor/depth_to_disparity_factor: 47.3
 * /elevation_mapping/input_sources/zed/sensor_processor/ignore_points_above: 1.4
 * /elevation_mapping/input_sources/zed/sensor_processor/lateral_factor: 0.001376915
 * /elevation_mapping/input_sources/zed/sensor_processor/p_1: 0.03287
 * /elevation_mapping/input_sources/zed/sensor_processor/p_2: -0.0001276
 * /elevation_mapping/input_sources/zed/sensor_processor/p_3: 0.485
 * /elevation_mapping/input_sources/zed/sensor_processor/p_4: 399.1046
 * /elevation_mapping/input_sources/zed/sensor_processor/p_5: 6.735e-06
 * /elevation_mapping/input_sources/zed/sensor_processor/type: stereo
 * /elevation_mapping/input_sources/zed/sensor_processor/voxelgrid_filter_size: 0.1
 * /elevation_mapping/input_sources/zed/topic: /camera/depth/poi...
 * /elevation_mapping/input_sources/zed/type: pointcloud
 * /elevation_mapping/length_in_x: 6.0
 * /elevation_mapping/length_in_x_init_submap: 1.0
 * /elevation_mapping/length_in_y: 6.0
[... 79 lines cut ...]
[ INFO] [1648663569.936265999]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1648663569.950289682]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1648663573.107441974]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1648663574.735791469, 0.001000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1648663574.747741709, 0.001000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1648663575.251642035, 0.001000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1648663575.279057755, 0.001000000]: Starting Laser Plugin (ns = /)
[ INFO] [1648663575.299410022, 0.001000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1648663575.337552671, 0.001000000]: Physics dynamic reconfigure ready.
[ INFO] [1648663575.378494502, 0.001000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1648663575.379390461, 0.001000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1648663575.382296886, 0.001000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1648663575.386357558, 0.001000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1648663575.390057284, 0.001000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1648663575.451213670, 0.001000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1648663575.466507272, 0.001000000]: DiffDrive(ns = //): Advertise odom on odom 
[spawn_urdf-3] process has finished cleanly
log file: /home/nbertsch/.ros/log/0ea855d8-b054-11ec-a556-1c1bb5e39bda/spawn_urdf-3*.log
[ INFO] [1648663581.451937249, 1.001000000]: Successfully launched node.
[ INFO] [1648663666.591700287, 22.692000000]: First corresponding point cloud and pose found, elevation mapping started. 
[ INFO] [1648663666.920748203, 22.777000000]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648663712.469113885, 32.801000000]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
[ INFO] [1648663761.629760742, 42.864000000]: Waiting for tf transformation to be available. (Message is throttled, 10s.)
```
Looks like it may be a TF issue from the ZED camera? Any idea how this could be fixed?

maximilianwulf, 2022-05-04:
> Waiting for tf transformation to be available. (Message is throttled, 10s.)

My best guess would be that the ZED driver does not properly publish the static tf tree. Sorry I have no experience with that driver.

hsurya08, 2023-04-16:
Hi, I am facing the same issue @nbertschmann. Are you able to find the solution?