ANYbotics/elevation_mapping #210: path planning

wxk514, 2022-05-06:
Hi, thank you very much for your work, do you have any suggestions on how to do robot path planning based on the traversability_estimation package, such as how to combine the ros navigation package.
thanks.

wxk514, 2022-05-08:
Hey, I save the map as a bag file through the following command. 

`  rosservice call /traversability_estimation/save_traversability_map_to_bag "file_path: '/home/user/your_bag.bag' topic_name: 'traversability_map_topic_name'"`
Is there any way to convert it into a yaml format map file? Thank you!

wxk514, 2022-05-09:
Hi, I save the traversable map through the following command
`rosrun map_server map_saver --occ 90 --free 10 -f mymap map:=/traversability_map_visualization/traversability_map
`
however,  the saved map data seems to be empty. What should I do? Thank you
[image]

xovobobo, 2022-10-13:
First you need to convert  /traversability_map_visualization/traversability_map to OccupancyGrid.msg