ANYbotics/elevation_mapping #214: How to accelerate update of elevation map

kosei1515, 2022-06-01:
Hi, I'm now using this repository. I changed these following parameters and now the fastest average of updates was 4s per 1 update.
```
- sensor_processor       apply_voxelgrid_filter:    true
                                     voxelgrid_filter_size:     0.03
- resolution: 0.03→This is at least necessary
- fused_map_publishing_rate: 3.0
- min_update_rate: 5.0
- num_callback_threads: 10
- postprocessor_num_threads: 10
- scanning_duration: 0.033
- mahalanobis_distance_threshold: 2.5
- multi_height_noise: 0.009
```
I want to accelerate it by parameter turning. Are there any ideas about any parameters for this problem? Or are there other solutions?
The following is my environment.
- PC: jetson NX
- ROS: ros melodic
- stereo camera: ZED mini

 Thank you for your help.

kosei1515, 2022-06-02:
> If you want to use GPU I can recommend https://github.com/leggedrobotics/elevation_mapping_cupy
Thank you.

> How did you measure it?

https://user-images.githubusercontent.com/81236654/171619044-82a2cb62-0482-439d-8986-dbd889600233.MOV

I saw ```/elavation_map``` topic in rviz. This is the video.

maximilianwulf, 2022-06-02:
Are we looking at the fused map or the raw map? 

Can you check the raw map?

kosei1515, 2022-06-06:
This is the raw map.
I think it is not so fast too.

https://user-images.githubusercontent.com/81236654/172095180-f9e67a8c-51e8-459e-a45e-18ebddbf86e5.mov

maximilianwulf, 2022-06-17:
Can you do

```
rostopic hz /your/topic
```

Did you compile with `release` flags?

2361098148, 2022-10-07:
hi， could you share how to use zed camera in elevation mapping? Im struggling. I d appreciate if you can help

kosei1515, 2022-10-08:
> Can you do
> 
> ```
> rostopic hz /your/topic
> ```
> 
> Did you compile with `release` flags?

I compiled with release flag, and it worked. 
Thank you for your help!

kosei1515, 2022-10-08:
> hi， could you share how to use zed camera in elevation mapping? Im struggling. I d appreciate if you can help

What is the problem? Please tell me the detail.
Could you compile the package?