ANYbotics/elevation_mapping #215: Map wraps around itself

mhowe8, 2022-06-13:
I am a university student using your elevation_mapping package for our robot. I am running the turtlebot demo in elevation_mapping and I wrote a program that prints the elevation map being read in real time. My main issue is right now is that the points of the point cloud that make contact beyond the edges of the elevation map wrap around itself, and I am wondering if I need to move the grid map frame or the grid map to a new position.
[image]

The points represent the point cloud and  the 'n's represent nan values. As you can see, the left side of the point cloud bleeds over to the right. When I spin the robot left 90 degrees, this is what I get:
[image]

I understand the map is stored as a circular buffer; is the fact that I am printing the map is what the cause of this issue is in the first place? Because of this, I don't see how I would successfully generate a cost map for path planning when the map wraps around.
Thank you.

maximilianwulf, 2022-06-15:
Hey, yes, do `map.convertToDefaultStartIndex();` before the other operations. That should solve your issue, for instance see https://github.com/ANYbotics/grid_map/blob/master/grid_map_filters/src/BufferNormalizerFilter.cpp#L27

mhowe8, 2022-06-15:
That worked perfectly! Thank you so much!

closed 2022-06-15 by mhowe8