ANYbotics/elevation_mapping #217: The point cloud from a depth camera vibrates as the robot moves

Yaru-Gu, 2022-06-22:
Hello,

I'm implementing this package with a quadruped robot. I'm using it in the Gazebo simulation. The input source is point clouds generated by a depth camera. Here are my **camera configurations in the robot description files**:
[image]

[image]

**My elevation map configurations:**
[image]

The problem is that the point cloud keeps vibrating when the robot moves in Gazebo, as shown in the following video. This vibration has given me inaccurate mapping. 

https://user-images.githubusercontent.com/96481145/175096270-30c6febf-2567-4818-a45a-f9fd2df956d6.mp4

 I couldn't figure out what causes the vibration. I appreciate your help.

maximilianwulf, 2022-06-23:
Is the pose estimation smooth?

Yaru-Gu, 2022-06-23:
> 

I think the pose estimation works fine. I used "rtabmap_ros" to publish poses at 5.0 hz.

Yaru-Gu, 2022-07-04:
Hi @maximilianwulf, regarding this issue, I tried to search for answers from some other examples. I found in this[ repository](https://github.com/chvmp/champ), that the point cloud obtained from the depth camera is also vibrating with the robot moving. I was thinking maybe this is normal? I wonder if you could check if this happens with your quadruped robot using the present elevation mapping package. I apologize in advance if this is an inappropriate request.

maximilianwulf, 2022-07-04:
Can you maybe upload a gif? The video's are corrupt on my machine.

Yaru-Gu, 2022-07-04:
> Can you maybe upload a gif? The video's are corrupt on my machine.

Sorry about that. Please the following gif files.
[image]
[image]

maximilianwulf, 2022-07-05:
Ah, now I can see it. That looks like the state estimation of your robot is jittery. 

Elevation mapping just subscribes to the information and does its calculations. I would recommend fixing the state estimation, instead of trying to fix it within elevation mapping.