ANYbotics/elevation_mapping #231: elevation map with high latency

skywalker-dell, 2022-11-10:
Hi. It seeded that the elevation map has high latency. I subscribed to the elevation map topic, and when there is a new message, I immediately calculated the latency as follows.
```
  auto now = ros::Time::now();
  auto latency = (now - grid_map_ptr->info.header.stamp).toSec();
```
the latency is about 3s on i9-12900KF and 12s on i7-1165G7.
I just simply modified the demo file to make it receive point clouds from the Kinect camera in the gazebo, the parameters are almost all the same, size(2.5, 2.5), resolution(0.05), etc.

skywalker-dell, 2022-11-10:
Another weird thing is that no matter how to tune those parameters like minimizing absolution or decreasing map size, the latency 
barely changes.

skywalker-dell, 2022-11-14:
It seemed that I need to downsample the point clouds

closed 2022-11-14 by skywalker-dell

Muhammad540, 2024-07-19:
> It seemed that I need to downsample the point clouds

@skywalker-dell  what approach are you using for downsampling the point cloud data ?