ANYbotics/elevation_mapping #238: Why do stairs in the map have a smooth transition?

dstx123, 2023-04-14:
I try to use this map in a quadruped robot for climbing stairs, but the accuracy of the map does not seem to be as high as that in the paper, existing a smooth transition of steps like this.
[image]
The quadruped robot seems unable to release a TF node called base_footprint. So I choose trunk instead of base_footprint. Is this going to be a problem?
Besides, I try to use the turtlebot3 demo to generate the map. The question I raised above still remain. 
[image]
How can I achieve the same effect as in the paper. Hope for your help!
[image]

dstx123, 2023-04-26:
> MeanInRadiusFilter

Could you teach me how to disable the MeanInRadiusFilter? I only find the "class MeanInRadiusFilter" in MeanInRadiusFilter.hpp. And I didn't find it in "postprocessor_pipeline.yaml". Hope for your help!

ZhangLiang-Cencen, 2023-05-01:
https://github.com/ANYbotics/elevation_mapping/tree/master/elevation_mapping_demos/config/postprocessing

nanbwrn, 2023-10-18:
May I ask for the steps to integrate the Go1 robot with this project to achieve a simulation result similar to yours, @dstx123

dstx123, 2023-10-18:
> May I ask for the steps to integrate the Go1 robot with this project to achieve a simulation result similar to yours, @dstx123

You can refer to the demo provided by this repository. This can be achieved by subscribing to the topic.

dstx123, 2023-10-18:
> https://github.com/ANYbotics/elevation_mapping/tree/master/elevation_mapping_demos/config/postprocessing

Thanks for your help. Sorry for forgetting to get back to you. I'm still curious though, is this achieved by adding some code to the "postprocessor_pipeline.yaml"? I didn't find anything about MeanInRadiusFilter in this file.