ANYbotics/elevation_mapping #244: How to save elevation map?

Aishkrish18, 2023-10-05:
Hello, 
Thank you for the interesting work. I would like to know how to save the elevation map using the rosservice call save_map. Every time i try it , it says processfile missing, I'm not sure how to solve this error. Can someone tell me how to save the map and load the map again to visualize on rviz and use?

Aishkrish18, 2023-10-05:
> Hello, Thank you for the interesting work. I would like to know how to save the elevation map using the rosservice call save_map. Every time i try it , it says processfile missing, I'm not sure how [...]

I figured out how to save it, thank you!

closed 2023-10-05 by Aishkrish18

jsj441805126, 2024-04-23:
> Hello, 你好， Thank you for the interesting work. I would like to know how to save the elevation map using the rosservice call save_map. Every time i try it , it says processfile missing, I'm not sure how to solve this error. Can someone tell me how to save the map and load the map again to visualize on rviz and use?感谢您所做的有趣的工作。我想知道如何使用 rosservice 调用 save_map 保存高程图。每次我尝试它时，它都会说进程文件丢失，我不知道如何解决这个错误。有人可以告诉我如何保存地图并再次加载地图以在 rviz 上可视化并使用吗？

Hello, I also met this problem, how do you solve it, please give me some guidance, will be grateful