ANYbotics/elevation_mapping #247: Orientation Issue in Elevation Mapping

MonsterQhz, 2023-10-27:
Hello developer,
I'm facing an issue and I'm seeking your assistance. I've subscribed to the /point_cloud and /tf topics and have successfully built an elevation map. However, there's an issue with the orientation of my grid map; it doesn't align with the vehicle's forward direction. I'm using tf data obtained from a SLAM mapping method, which is relatively accurate.
Could you please provide some insights into what might be causing this issue? And, do you have any recommendations on how I can resolve it? I've attached an image for reference.
Thank you for your assistance.
[image]

Best regards.

lichao951787328, 2023-10-30:
I have the same problem. The Orientation  align with the initial orientation, so the current submap is not align with current base?

MonsterQhz, 2023-11-01:
> I have the same problem. The Orientation align with the initial orientation, so the current submap is not align with current base?

Have you solve the problem? I try to revise the frameId, but it doesn't work. And I guess that the setting of grid map maybe cause the problem.

lichao951787328, 2023-11-01:
I believe it is because of the code's own configuration that requires the map to have the same orientation as the initial direction. You can take a look at the 'grid_map' repository, where the method for obtaining subregions does not include direction transformation.

lichao951787328, 2023-11-12:
I have a solution to this problem. You can use lineIterator to get a tilted submap aligned with the current orient of your vehicle.

MonsterQhz, 2023-11-12:
Thank you! Could you provide a more detailed description of your strategy? Here are my views and proposed solutions.I believe the issue is caused by incorrect calculation of the transformation matrix (tf) on my part. I used a combination navigation system to calculate the pose transformation of the vehicle, but I did not compensate for the heading angle. When I attempted to compensate for the heading angle, the orientation became correct, but the height values in the map were incorrect. I am working to resolve this issue. I think there are two possible ways to address this problem: 1. Utilize other sensors and methods to obtain the pose, such as when I use an IMU, this issue does not occur. 2. Refactor the code and use the grid_map library to build the elevation map myself. I discovered that it is based on map-based descriptions, can describe the current frame point cloud, and consider information from adjacent frames. Of course, this strategy of trading space for time can reduce processing time. The core of its code is to calculate the elevation information of the grid through Mahalanobis distance, and we can also use its move function when refactoring our code.

lichao951787328, 2023-11-12:
My suggestion is that:  since the constructed local map must be consistent with the local coordinate system, you can calculate the elevation map at the current time based on the previous local elevation map. The value of each cell in the elevation map at the current time can be calculated based on the value of the previous elevation map and the transformation matrix between the two times.

Additionally, if you want to construct submap at any angle, you can extend the lineIterator. If you have any questions, please feel free to contact me again.

                       
Original Email
                       
                     

Sender:"MonsterQhz"< ***@***.*** &gt;;

Sent Time:2023/11/12 19:16

To:"ANYbotics/elevation_mapping"< ***@***.*** &gt;;

Cc recipient:"lichao951787328"< ***@***.*** &gt;;"Comment"< ***@***.*** &gt;;

Subject:Re: [ANYbotics/elevation_mapping] Orientation Issue in ElevationMapping (Issue #247)

 
Thank you! Could you provide a more detailed description of your strategy? Here are my views and proposed solutions.I believe the issue is caused by incorrect calculation of the transformation matrix (tf) on my part. I used a combination navigation system to calculate the pose transformation of the vehicle, but I did not compensate for the heading angle. When I attempted to compensate for the heading angle, the orientation became correct, but the height values in the map were incorrect. I am working to resolve this issue. I think there are two possible ways to address this problem: 1. Utilize other sensors and methods to obtain the pose, such as when I use an IMU, this issue does not occur. 2. Refactor the code and use the grid_map library to build the elevation map myself. I discovered that it is based on map-based descriptions, can describe the current frame point cloud, and consider information from adjacent frames. Of course, this strategy of trading space for time can reduce processing time. The core of its code is to calculate the elevation information of the grid through Mahalanobis distance, and we can also use its move function when refactoring our code.

MonsterQhz, 2023-11-14:
I've solved my problem, thanks for everyone's help. If your lidar is not installed horizontally, please rotate it to level. I will close this issue.

closed 2023-11-14 by MonsterQhz

Lkaho, 2023-11-14:
Could you please explain the method for aligning the orientation of the elevation map with that of the vehicle? Thanks!

lichao951787328, 2023-11-14:
The key is to calculate the positions of the four points in the rectangle, and then combine them with linriterator

                       
Original Email
                       
                     

Sender:"Lkaho"< ***@***.*** &gt;;

Sent Time:2023/11/14 21:00

To:"ANYbotics/elevation_mapping"< ***@***.*** &gt;;

Cc recipient:"lichao951787328"< ***@***.*** &gt;;"Comment"< ***@***.*** &gt;;

Subject:Re: [ANYbotics/elevation_mapping] Orientation Issue in ElevationMapping (Issue #247)

 
Could you please explain the method for aligning the orientation of the elevation map with that of the vehicle? Thanks!