ANYbotics/elevation_mapping #249: migration to ros2 humble?

er1c-cartman, 2023-12-06:
Hi. first of all, thanks for sharing your works
I've tried elevation mapping with traversability estimation in ROS1 noetic environment, and it works well
Now I'm trying to use in ROS2 humble environment, and i guess the ros2 branch cannot be built due to the issue that ros2 humble doesn't use catkin at all
Is there any plans or some works on package that available to use in ROS2 humble environment?
best regards

mmiles19, 2024-01-15:
I am also interested in using an updated `elevation_mapping` package which supports ROS2 Humble+. 
It seems this repo is still actively maintained. Are the developers interested in community conbributions? @PrajishKumar (pleased to make your acquaintance - pinging you since you had the most recent commit in `master`)

PrajishKumar, 2024-01-16:
> I am also interested in using an updated `elevation_mapping` package which supports ROS2 Humble+. It seems this repo is still actively maintained. Are the developers interested in community [...]

@Magnusgaertner @maximilianwulf, do you have plans to upgrade this to ROS2, or accept PRs?

maximilianwulf, 2024-01-16:
Hi all, at some point we created a ros2 branch that the open source community could work on. https://github.com/ANYbotics/elevation_mapping/tree/ros2. I would recommend check-in out that one, but ANYbotics is currently not actively maintaining that branch. I hope that helps.