ANYbotics/elevation_mapping #256: Some Info needed for building elevation_mapping in ROS2 Humble.

babahadjsaid, 2024-02-12:
Hello,
So I'm trying to fix the bugs i'm getting when building this lib in ros2 humble, I'm not using your repo as you don't provide Ros2 implementation, i'm trying to build this one , i had some issues concerning a couple of lines, some of them were easy and strainght forward, but some (the ones i will share here), were a bit tricky, in sense that they need some technical detail, and they are:
in this line:
https://github.com/adyczech/elevation_mapping/blob/b2481c22100d83956483dce7db7d62c459db0736/elevation_mapping/src/sensor_processors/SensorProcessorBase.cpp#L136
the Duration class was init with a value of 1.0, which doesn't meet any constractor looking to the constractors i found that the way to do so is by specifying the seconds and nanoseconds as integer, So my question is what is the value you meant by 1.0, is it 1 sec ?

whereas in this line:
https://github.com/adyczech/elevation_mapping/blob/b2481c22100d83956483dce7db7d62c459db0736/elevation_mapping/src/input_sources/Input.cpp#L28
a some parameters were being declared, but the syntax needed a default value if the user didn't provide one, can you give me  the list of default values that i can put there ?
Thank you so much.

Muhammad540, 2024-07-03:
hey @babahadjsaid , did you manage to successfully port the package to ROS2 humble ?

babahadjsaid, 2024-07-03:
yes i did. here it is.
https://github.com/babahadjsaid/elevation_mapping