ANYbotics/elevation_mapping #257: leg filterig for quadruped robots (in the ele_map legs are mapped as well)

rushelby123, 2024-03-01:
The problem is that in the elevation mapping are mapped also the legs besides the environment, as a result the elevation mapping show an undesired mapping around the robot and i would like to correct this problem. I have seen that there is a service called masked_replace, there is any chance to use it to solve my problem and is it able to solve it dynamically? if not how what is your suggestion to solve the issue? could you provide some example to begin with?

maximilianwulf, 2024-03-01:
Hey, elevation mapping currently does not provide the functionality to filter out the robot body dynamically. 

It is recommended to filter out the missing parts before fusing it into map.

rushelby123, 2024-03-01:
Hi thanks for the answer, what are your suggestion in this matter? I never approached a problem like this so i was looking to find an easy way to solve it. I found two possible solutions: 
The first approach consists in creating a mask that is updated along with the tf, in this way i may be able to neglect the points of the point cloud that belongs to the robot, but it gets hard to understand how the datastucture works and how to update the mask along with the tf.
The second approach may be easier but i think it would result poorly in the final elevation mapping, is to use a gaussian like filter in the neighbourhood of the feet so at the end whatever is under the robot will be filtered.
What do you think about these two solutions? I think the first approach would be the best way to solve it but I have not that much experience with ROS to solve it in a fast way. Thank you in advance

rushelby123, 2024-03-01:
[image]