ANYbotics/elevation_mapping #266: Quick Port of Elevation Mapping to ROS2 Humble

Muhammad540, 2024-07-19:
Hi,

I have done a quick port of the elevation mapping package to ROS2 Humble. The ported version is available in my repository https://github.com/Muhammad540/elevation_mapping.git

I believe this will be beneficial to the community, if you have any questions feel free to ask. I want to thank the authors of the elevation_mapping package for their excellent work.

RMichaelSwan, 2024-10-17:
It looks like @Muhammad540 manually forked https://github.com/adyczech/elevation_mapping.git and mangled it a bit? That's a bit peculiar.

Muhammad540, 2024-10-17:
> It looks like @Muhammad540 manually forked https://github.com/adyczech/elevation_mapping.git and mangled it a bit? That's a bit peculiar.

Thanks for pointing that out! Yes, I did a manual fork the elevation_mapping repository and made some changes to fit the needs of my project. I understand the modifications might look a bit different from the original, but they were necessary for what I was working on. If you’re interested, I’d be happy to explain the changes in more detail or share how they fit my use case.

closed 2024-10-17 by Muhammad540