ANYbotics/elevation_mapping #267: Little elevation map in velodyne 16 lidar!

KJ-Falloutlast, 2024-09-06:
Really thankful for your masterpiece, but I have a problem: 
I've already set my urdf and config file as the tutorial did as following, but I can't not get the elevator map. please help me with this problem, and I would be appreciate!
* this is my console output:
[image]
* this is my config:
```yaml
input_sources:                                                                  
  front:                                                                        
    type: pointcloud                                                            
    topic: /velodyne_points/points_downsampled                               
    queue_size: 10           #queue_size 必须大于5才行                                                  
    publish_on_update: true                                                     
    sensor_processor:                                                           
      type: laser                                                               
      #ignore_points_above: 0.0                                                 
      #ignore_points_below: -4.0                                                
      min_radius: 0.018                                                         
      beam_angle: 0.0006                                                        
      beam_constant: 0.0015                                                     
map_frame_id: "odom"                                                             
robot_base_frame_id: "base_link"                                                
robot_pose_with_covariance_topic: "/base_link_pose"                             
robot_pose_cache_size: 200                                                      
track_point_frame_id: "base_link"  
length_in_x: 5                                                                     
length_in_y: 5                                                                     
position_x: 0.0                                                                    
position_y: 0.0                                                                    
resolution: 0.08                                                                   
min_variance: 0.00009                                                              
max_variance: 0.01                                                                 
mahalanobis_distance_threshold: 2.5                                                
multi_height_noise: 0.00002
sensor_frame_id: "velodyne" #velodyne
```
* this is my tf_tree
[image]

KJ-Falloutlast, 2024-09-06:
[image]
the map is just like this, which is so weird!

rolun61, 2025-07-29:
Not sure if you have solved this issue already.
But perhaps you could try moving the model car around and see if the point cloud can gather information about its surroundings and see if that improves the situation you're experiencing.